MaslowCNC/ESP3D-WEBUI #201: [FEATURE REQUEST] bit change height

davidelang, 2025-09-21:
add a numeric config variable bitChangeHeight in preferences.yaml defaulting to 60
add a button the the UI that when pressed, stores the current Z height, moves up to this height (in mm) and changes from 'change bit' to 'probe for bit length' if a probe is enabled, or 'lower bit' that returns the Z to the prior height if the probe is not enabled.

davidelang, 2025-10-19:
replaced by #233 due to copilot hitting a limit

closed 2025-10-19 by davidelang

BarbourSmith, 2025-10-24:
@davidelang Why is this better than using the existing z-axis jog buttons? This feels like adding complexity where we aren't forced to

davidelang, 2025-10-24:
BarbourSmith wrote:

> This is a great point, but I think that the right way to solve it is with a 
> soft limit that prevents us from going off the top. That would solve the issue 
> without adding any complexity to the user experience, right?

not when the soft limit triggers and error (doesn't that cause the rehang 
dance??)

This also moves you back down to where you were (or starts probing, which should 
be on this screen anyway rather than having to go to a different screen)

With this button, a bit change is a button press, change the bit, and a button 
press (possibly with manual z-home setting)

without this button, but with soft limits, you either need to hit Z jog a lot, 
or you have to change the jog distance, jog up, then

if no probe: change the jog distance and jog down, change the jog disatance and 
fine tune the z-home settings

If probe: switch to a different screen and start the Z probe cycle

This is far simpler for the user, in exchange for one button that they are not 
required to use.

David Lang

davidelang, 2025-10-24:
BarbourSmith wrote:

> BarbourSmith left a comment (MaslowCNC/ESP3D-WEBUI#201)
>
> I think that the correct soft limit behavior should be to just not let it go any higher. Like if we type in 1000 and press z-up it should just to to the top and stop there

you still need to either hit the up button a lot or change the distance each 
button press moves you to go up, then to go down, then to fine tune

but the default jog behavior of FluidNC is that if you are instructed to jog 
beyond the soft limit, it aborts the movement, not moves to the limit.

David Lang

BarbourSmith, 2025-10-24:
I'm not saying that it's a bad idea, but feature bloat happens because individually each feature makes sense, but then taken all together the software has so many features that it's hard for users to know how to use them.